hideakitai/ESP32DMASPI #2: SPI slave sends data using SPI_MODE3. Master receives data incorrectly.

hu-Eric, 2021-03-17:
SPI slave is initialized to SPI_MODE2, and master receives the data correctly

hideakitai, 2021-03-17:
@hu-Eric Which code are you talking about?

closed 2021-03-25 by hideakitai

hu-Eric, 2021-03-26:
// make sure to use DMA buffer
    if ((dma_chan != 1) && (dma_chan != 2))
    {
        printf("[WARNING] invalid DMA channel %d, force to set channel 2. make sure to select 1 or 2\n", dma_chan);
        dma_chan = 2;
    }
    if ((mode != SPI_MODE1) && (mode != SPI_MODE3))
    {
        printf("[WARNING] invalid SPI channel %d, force to set SPI_MODE3. make sure to select MODE1 or MODE3 (because of DMA)\n", mode);
        mode = SPI_MODE3;
    }
Why set it to SPI_MODE3? The receiving data sent to the master device is different from the data sent. Setting it to SPI_MODE2 can get the correct data.

hideakitai, 2021-03-26:
Please see https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/spi_slave.html#restrictions-and-known-issues

If you still have the problem, please let me know ide version, esp32 version, library version, minimum code which can reproduce it, and schematic(connection).