hideakitai/ESP32DMASPI #3: is master write blocking until all buffer sent?

mark-hahn, 2021-06-02:
Is master write blocking until all buffer sent?   Isn't the point of dma to be able to use it in background while doing other things?

hideakitai, 2021-06-03:
@mark-hahn Hi, you can use non-blocking send by `queue()`

mark-hahn, 2021-06-03:
Thanks -- I missed that.

hideakitai, 2021-06-03:
Maybe I should improve README and examples.

closed 2021-06-03 by hideakitai

brahma-dev, 2021-10-13:
@hideakitai It seems like queue() only works if I yield() at some point. I am very new to this so probably doing something wrong.

hideakitai, 2021-10-14:
@brahma-dev @mark-hahn  Ah, sorry for that. You should wait until the transaction ends when using `transfer()` or `queue()` + `yield()`. No API can complete transactions asynchronously (in the background).

See the official API reference for more details.
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/spi_master.html#api-reference-spi-master