hideakitai/ESP32DMASPI #4: For every 1 master transaction I got 8-10 slave transactions

bakwc, 2021-06-13:
_Master code_:
```
int n = 0;
void loopMaster() {
    n += 1;
    spi_master_tx_buf[0] = n % 100;

    master->transfer(spi_master_tx_buf, spi_master_rx_buf, BUFFER_SIZE);

    printf("RX BUF[0] = %d\n", (int)spi_master_rx_buf[0]);

    delay(1000);
}
```

_Slave code_:

```
void loopSlave() {
    // set buffer (reply to master) data here

    spi_slave_tx_buf[0] = (spi_slave_rx_buf[0] * 2) % 100;

    // if there is no transaction in queue, add transaction
    if (slave->remained() == 0)
        slave->queue(spi_slave_rx_buf, spi_slave_tx_buf, BUFFER_SIZE);

    // if transaction has completed from master,
    // available() returns size of results of transaction,
    // and buffer is automatically updated

    while (slave->available())
    {
        printf("Request from master: %d\n", (int)spi_slave_rx_buf[0]);
        slave->pop();
    }
}
```
`setup` is the same as in your README example

Slave output:

```
20:35:39.502 -> Request from master: 164
20:35:43.576 -> Request from master: 2
20:35:43.576 -> Request from master: 2
20:35:43.576 -> Request from master: 2
20:35:43.576 -> Request from master: 2
20:35:43.576 -> Request from master: 8
20:35:43.576 -> Request from master: 8
20:35:43.576 -> Request from master: 8
20:35:43.576 -> Request from master: 8
20:35:43.576 -> Request from master: 8
20:35:43.576 -> Request from master: 8
20:35:44.583 -> Request from master: 4
20:35:44.583 -> Request from master: 4
20:35:44.583 -> Request from master: 4
20:35:44.583 -> Request from master: 4
20:35:44.583 -> Request from master: 4
20:35:44.583 -> Request from master: 4
20:35:44.583 -> Request from master: 4
20:35:44.583 -> Request from master: 4
20:35:44.583 -> Request from master: 141
20:35:44.583 -> Request from master: 103
20:35:45.586 -> Request from master: 103
20:35:45.586 -> Request from master: 103
20:35:45.586 -> Request from master: 103
20:35:45.586 -> Request from master: 103
20:35:45.586 -> Request from master: 103
20:35:45.586 -> Request from master: 103
20:35:45.586 -> Request from master: 103
20:35:45.586 -> Request from master: 103
20:35:45.586 -> Request from master: 103
20:35:46.582 -> Request from master: 8
20:35:46.582 -> Request from master: 8
20:35:46.582 -> Request from master: 143
20:35:46.582 -> Request from master: 143
20:35:46.582 -> Request from master: 143
20:35:46.582 -> Request from master: 143
20:35:46.582 -> Request from master: 143
20:35:46.582 -> Request from master: 143
20:35:46.582 -> Request from master: 143
```

_Expected behavior_: got 1 printf on slave per second
_Actual behavior_: got 8-10 printf on slave per second

hideakitai, 2021-06-14:
Hi @bakwc , thank you for reporting.

Have you tried [master_simple](https://github.com/hideakitai/ESP32DMASPI/blob/master/examples/master_simple/master_simple.ino) and [slave_simple_polling](https://github.com/hideakitai/ESP32DMASPI/blob/master/examples/slave_simple_polling/slave_simple_polling.ino) examples? Could you try if they work and let me know about your environment, please?

If they won't work, please give me your minimum reproducible whole code.

bakwc, 2021-06-14:
I tried them too -> same behavior.
I receive 0 0 0 0 0 0 0 0 ... on slave, 5 time per second.
The code is excatly master_simple and slave_simple_polling.
Board - esp32cam.

bakwc, 2021-06-14:
It was a hardware issue, i used 20cm wires with plastic connectors (no soldering).
I took 3cm wires and soldered them, and everything works without any issues.
Thank you for support and for the library!

closed 2021-06-14 by bakwc

bakwc, 2021-06-15:
I faced another issue, all my data is shifted by 1 bit. Eg, when I sending:
`0 1 2 3 4 5 6 7 8 9 10 11 12`
Slave get:
`0 2 4 6 8 10 12 14 16 18 20 22 24`
I tried different frequencies, tried to use SPI_MODE1 and SPI_MODE3.