hideakitai/ESP32DMASPI #5: Shifted bit when using SPI_MODE1 or SPI_MODE3

bakwc, 2021-06-15:
With SPI_MODE0 - everything works fine, but there is a warning in logs.
Despite of of the warning - SPI_MODE0 works good. I suggest to remove warning and allow to use SPI_MODE0.

Looks like MODE0 works with DMA: https://github.com/espressif/esp-idf/commit/58955a79a27d3c7331eaec6e464878df42615a36

I was unable to make other modes working.

hideakitai, 2021-06-16:
@bakwc Thanks for reporting. What is the difference of settings from #4? Please let me know which change caused it. 

I will remove the warning because the limitation does not depend on this library but [esp-idf](https://github.com/espressif/esp-idf/search?q=dma+spi+bit+shift&type=issues) (we can find many issues about it).

hideakitai, 2021-06-16:
194d62783fe244fd1d67e621068eeb9294e9127c

bakwc, 2021-06-16:
Thank you. For me it is 100% reproducible when connecting esp32cam SPI master with esp32cam SPI slave.

hideakitai, 2021-06-21:
@bakwc I've removed the warning and updated README about bit shifting. Close this issue because this is not the problem of this library. Thank you for reporting.

closed 2021-06-21 by hideakitai