hideakitai/ESP32DMASPI #6: Simple Slave example for Ping-Pong Game?

openedhardware, 2021-07-23:
Thanks for this amazing library!

I just wonder if you could provide a simple slave example that returns the received bytes back to the master?

hideakitai, 2021-07-24:
Hi, you can realize it by copying received bytes to tx buffer. It will be transferred in the next transaction.

For example, based on this example,

https://github.com/hideakitai/ESP32DMASPI/blob/master/examples/slave_simple_polling/slave_simple_polling.ino

You can do it by changing these lines

https://github.com/hideakitai/ESP32DMASPI/blob/dca4a8b0ea53d3eb6ab45d979ab1cf5bb2f53d38/examples/slave_simple_polling/slave_simple_polling.ino#L50-L58

to

```C++
    while (slave.available())
    {
        // show received data
        for (size_t i = 0; i < BUFFER_SIZE; ++i)
            printf("%d ", spi_slave_rx_buf[i]);
        printf("\n");

        // copy received data to tx buffer for next transfer
        memcpy(spi_slave_tx_buf, spi_slave_rx_buf, BUFFER_SIZE);

        slave.pop();
    }
```

openedhardware, 2021-07-24:
Perfect, thanks!

closed 2021-07-24 by openedhardware