hideakitai/ESP32DMASPI #10: Communications Between 3 uC

fteloken, 2021-10-17:
Hello, first thanks for you library, this is simple and objective.
I have some questions and problems about with a particular implementation.
I have test between two devices firstly, but the data receive in slave device is correctly, but in master request this data is changing from a constant value.
The code Master Device appears like:
`#include <ESP32DMASPIMaster.h>

ESP32DMASPI::Master master;
static const uint8_t BUFFER_SIZE = 255;
uint8_t *spi_master_tx_buf;
uint8_t *spi_master_rx_buf;

void setup()
{
    Serial.begin(115200);
    // to use DMA buffer, use these methods to allocate buffer
    spi_master_tx_buf = master.allocDMABuffer(BUFFER_SIZE);
    spi_master_rx_buf = master.allocDMABuffer(BUFFER_SIZE);

    master.setDataMode(SPI_MODE0);
    master.setFrequency(SPI_MASTER_FREQ_8M);
    master.setMaxTransferSize(BUFFER_SIZE);
    master.setDMAChannel(1); // 1 or 2 only
    master.setQueueSize(1);  // transaction queue size

    // begin() after setting
    // HSPI = CS: 15, CLK: 14, MOSI: 13, MISO: 12
    // VSPI = CS: 5, CLK: 18, MOSI: 23, MISO: 19
    master.begin(HSPI);
}

void loop()
{
    spi_master_tx_buf[0] = 55; // start and wait to complete transaction
    master.transfer(spi_master_tx_buf, spi_master_rx_buf, BUFFER_SIZE);
    Serial.print("RX BUF[0] = ");
    Serial.print(spi_master_rx_buf[0]);
    Serial.print(" | TC BUF[0] = ");
    Serial.print(spi_master_tx_buf[0]);
    Serial.println();
    vTaskDelay(pdMS_TO_TICKS(500));
}`

The code for Slave Device:
`#include <ESP32DMASPISlave.h>

ESP32DMASPI::Slave slave;

static const uint8_t BUFFER_SIZE = 255;
uint8_t *spi_slave_tx_buf;
uint8_t *spi_slave_rx_buf;

void setup()
{
    Serial.begin(115200);

    // to use DMA buffer, use these methods to allocate buffer
    spi_slave_tx_buf = slave.allocDMABuffer(BUFFER_SIZE);
    spi_slave_rx_buf = slave.allocDMABuffer(BUFFER_SIZE);

    slave.setDataMode(SPI_MODE0);
    slave.setMaxTransferSize(BUFFER_SIZE);
    slave.setDMAChannel(2); // 1 or 2 only
    slave.setQueueSize(1);  // transaction queue size

    // begin() after setting
    // HSPI = CS: 15, CLK: 14, MOSI: 13, MISO: 12
    // VSPI = CS: 5, CLK: 18, MOSI: 23, MISO: 19
    slave.begin(HSPI);
}

void loop()
{
    // set buffer (reply to master) data here
    spi_slave_tx_buf[0] = 88;
    // if there is no transaction in queue, add transaction
    slave.wait(spi_slave_rx_buf, spi_slave_tx_buf, BUFFER_SIZE); // Send data
    // if transaction has completed from master,
    // available() returns size of results of transaction,
    // and buffer is automatically updated
    while (slave.available())
    {
        Serial.print("Request from master [RX]: ");
        Serial.print(spi_slave_rx_buf[0]);
        Serial.print(" | Request from master [TX]: ");
        Serial.print(spi_slave_tx_buf[0]);
        Serial.println();
        slave.pop();
    }
}`

In master Device the Serial print this:

RX BUF[0] = 8 | TC BUF[0] = 55
RX BUF[0] = 88 | TC BUF[0] = 55
RX BUF[0] = 88 | TC BUF[0] = 55
RX BUF[0] = 44 | TC BUF[0] = 55
RX BUF[0] = 44 | TC BUF[0] = 55
RX BUF[0] = 88 | TC BUF[0] = 55
RX BUF[0] = 44 | TC BUF[0] = 55
RX BUF[0] = 88 | TC BUF[0] = 55
RX BUF[0] = 44 | TC BUF[0] = 55
RX BUF[0] = 88 | TC BUF[0] = 55
RX BUF[0] = 44 | TC BUF[0] = 55
RX BUF[0] = 44 | TC BUF[0] = 55
RX BUF[0] = 88 | TC BUF[0] = 55
RX BUF[0] = 44 | TC BUF[0] = 55
RX BUF[0] = 44 | TC BUF[0] = 55
RX BUF[0] = 88 | TC BUF[0] = 55
RX BUF[0] = 88 | TC BUF[0] = 55

I change SPI Mode to 4 modes, speed data, but this dont resolve the issue !?
The slave mode receive data correctly !

My second question is about increase a second slave with different SS control pinout, adding two devices in the system.
I can set manually the master.begin(HSPI), for differents objects ?
Thanks a lot !!

hideakitai, 2021-10-18:
@fteloken Hi, thanks for reporting. Do the examples work fine on your devices?

hideakitai, 2021-11-01:
I'm going to close this issue, but feel free to reopen if you still have a problem.

closed 2021-11-01 by hideakitai