hideakitai/ESP32DMASPI #12: TX data Slave->Master

Fasol0, 2022-01-04:
Hi,

I am reopening the issue in ESP32Slave here, in order to find a suitable workaround for slave mode:

- It seems that the best reliable communication for me is obtained when using SPI_MODE1 on master (which is an external device) and SPI_MODE3 on slave (esp32) but still Slave->Master communication does not work.
Can you confirm you do have a configuration which works reliably for Master->Slave AND Slave->Master communication?  maybe MODE0 with setDutyCyclePos(96) workaround ?

- If the Master is another device which is not an ESP32, should it work if I find an equivalent function to  setDutyCyclePos() ? 

Thank you for your help

hideakitai, 2022-01-06:
@Fasol0 

> Can you confirm you do have a configuration which works reliably for Master->Slave AND Slave->Master communication?

In the same ESP32 device (one device has two SPIs), the `master_slave_polling.ino` example works correctly in my environment.

> If the Master is another device which is not an ESP32, should it work if I find an equivalent function to setDutyCyclePos() ?

I don't know because I haven't tested. It may depend on the master device. Please try not using `setDutyCyclePos()` first (with default config).

This issue seems to be related to this problem.
https://github.com/hideakitai/ESP32DMASPI#restrictions-and-known-issues-for-spi-with-dma-buffer-help-wanted

I don't have enough time to analyze these issues, but it depends on the esp-idf version (the internal process of esp32). Arduino IDE + ESP32 v1.0.6 + ESP32DMASPI v0.1.2 works fine, but Arduino IDE + ESP32 v2.0.0 + ESP32DMASPI v0.12 doesn't work correctly. Because of the timing issues.

Also, refer to the documents about timing considerations.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html#transfer-speed-considerations
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_slave.html#speed-and-timing-considerations

hideakitai, 2022-03-07:
I am closing this issue because no further information has been provided. Please feel free to reopen this.

closed 2022-03-07 by hideakitai