hideakitai/ESP32DMASPI #13: ESP32DMASPI v0.2.0 : Compile Error ESP32DMASPISlave.h

JBSchueler, 2022-01-04:
I get a compile error
```
In file included from C:\projects\Arduino\libraries\ESP32DMASPI\examples\slave_simple_task\slave_simple_task.ino:1:0:
C:\projects\Arduino\libraries\ESP32DMASPI/ESP32DMASPISlave.h:50:5: sorry, unimplemented: non-trivial designated initializers not supported
     };
     ^
C:\projects\Arduino\libraries\ESP32DMASPI/ESP32DMASPISlave.h:50:5: sorry, unimplemented: non-trivial designated initializers not supported
C:\projects\Arduino\libraries\ESP32DMASPI/ESP32DMASPISlave.h:53:19: error: 'SPI_DMA_CH_AUTO' was not declared in this scope
     int dma_chan {SPI_DMA_CH_AUTO};  // must be 1, 2 or SPI_DMA_CH_AUTO
                   ^
C:\projects\Arduino\libraries\ESP32DMASPI/ESP32DMASPISlave.h:53:34: error: cannot convert '<brace-enclosed initializer list>' to 'int' in initialization
     int dma_chan {SPI_DMA_CH_AUTO};  // must be 1, 2 or SPI_DMA_CH_AUTO
                                  ^
C:\projects\Arduino\libraries\ESP32DMASPI/ESP32DMASPISlave.h: In member function 'void arduino::esp32::spi::dma::Slave::setDMAChannel(uint8_t)':
C:\projects\Arduino\libraries\ESP32DMASPI/ESP32DMASPISlave.h:188:24: error: 'SPI_DMA_CH_AUTO' was not declared in this scope
             dma_chan = SPI_DMA_CH_AUTO;
                        ^
Using library ESP32DMASPI at version 0.2.0 in folder: C:\projects\Arduino\libraries\ESP32DMASPI 
```

Fasol0, 2022-01-05:
Yes I had the same issues,

- You can fix the first one by adding:
#define SPI_DMA_CH_AUTO 3
if it does not work you can replace '3' (which is automatic channel selection) by '1' or '2'
in ESP32DMASPISlave.h

- You can fix the second one by explicitely initialize bus_cfg, for instance:
    spi_bus_config_t bus_cfg {
        .mosi_io_num = 13,        // HSPI
        .miso_io_num = 12,        // HSPI
        .sclk_io_num = 14,        // HSPI
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
        .max_transfer_sz = 4092,  // default: 4092 if DMA enabled, SOC_SPI_MAXIMUM_BUFFER_SIZE if DMA disabled
        .flags = SPICOMMON_BUSFLAG_SLAVE,
        .intr_flags = ESP_INTR_FLAG_IRAM
    };

you can see that compared to the original file, I added the missing fields .quadwp_io_num, .wuadhd_io_num (both  can be set to -1 without any effect) and .intr_flags (you can set it to 0 or ESP_INTR_FLAG_IRAM (not sure the latter is really effective though)

JBSchueler, 2022-01-05:
@Fasol0 , thanks.
I noticed `SPI_DMA_CH_AUTO `should have the value 3 by looking at the code
https://github.com/hideakitai/ESP32DMASPI/blob/9a77471f992a5573aaff1e2fb566df819637cb88/ESP32DMASPISlave.h#L184
It accepts 3 as well so that must be `SPI_DMA_CH_AUTO `

Nevertheless, it should be solved. Personally I don't push code which is does not compile.

hideakitai, 2022-01-06:
@JBSchueler @Fasol0 
Hi, v0.2.0 requires ESP32 version >= 2.0.0 for Arduino IDE, and not supported on PlatformIO now. Added supported version matrix to README

JBSchueler, 2022-01-06:
@hideakitai 
Thanks for the information, I was not aware of that.

closed 2022-01-06 by JBSchueler