hideakitai/ESP32DMASPI #14: ESP32DMASPI v0.2.0 : Compile Error ESP32DMASPIMaster.h

JBSchueler, 2022-01-04:
I get a compile error
```
In file included from C:\projects\Arduino\libraries\ESP32DMASPI\examples\master_slave_queue\master_slave_queue.ino:1:0:
C:\projects\Arduino\libraries\ESP32DMASPI/ESP32DMASPIMaster.h:51:5: sorry, unimplemented: non-trivial designated initializers not supported
     };
     ^
C:\projects\Arduino\libraries\ESP32DMASPI/ESP32DMASPIMaster.h:51:5: sorry, unimplemented: non-trivial designated initializers not supported
C:\projects\Arduino\libraries\ESP32DMASPI/ESP32DMASPIMaster.h:55:19: error: 'SPI_DMA_CH_AUTO' was not declared in this scope
     int dma_chan {SPI_DMA_CH_AUTO};  // must be 1, 2 or SPI_DMA_CH_AUTO
                   ^
C:\projects\Arduino\libraries\ESP32DMASPI/ESP32DMASPIMaster.h:55:34: error: cannot convert '<brace-enclosed initializer list>' to 'int' in initialization
     int dma_chan {SPI_DMA_CH_AUTO};  // must be 1, 2 or SPI_DMA_CH_AUTO
                                  ^
C:\projects\Arduino\libraries\ESP32DMASPI/ESP32DMASPIMaster.h: In member function 'void arduino::esp32::spi::dma::Master::setDMAChannel(uint8_t)':
C:\projects\Arduino\libraries\ESP32DMASPI/ESP32DMASPIMaster.h:291:24: error: 'SPI_DMA_CH_AUTO' was not declared in this scope
             dma_chan = SPI_DMA_CH_AUTO;
                        ^
C:\projects\Arduino\libraries\ESP32DMASPI/ESP32DMASPIMaster.h: In member function 'void arduino::esp32::spi::dma::Master::addTransaction(uint8_t, uint8_t, uint8_t, uint32_t, uint16_t, uint64_t, size_t, const uint8_t*, uint8_t*)':
C:\projects\Arduino\libraries\ESP32DMASPI/ESP32DMASPIMaster.h:332:43: error: 'SPI_TRANS_VARIABLE_DUMMY' was not declared in this scope
         transactions.back().base.flags |= SPI_TRANS_VARIABLE_DUMMY;
                                           ^
C:\projects\Arduino\libraries\ESP32DMASPI/ESP32DMASPIMaster.h:344:29: error: '__gnu_cxx::__alloc_traits<std::allocator<spi_transaction_ext_t> >::value_type {aka struct spi_transaction_ext_t}' has no member named 'dummy_bits'
         transactions.back().dummy_bits = dummy_bits;
                             ^
In file included from C:\projects\Arduino\libraries\ESP32DMASPI\examples\master_slave_queue\master_slave_queue.ino:2:0:
C:\projects\Arduino\libraries\ESP32DMASPI/ESP32DMASPISlave.h: At global scope:
C:\projects\Arduino\libraries\ESP32DMASPI/ESP32DMASPISlave.h:50:5: sorry, unimplemented: non-trivial designated initializers not supported
     };
     ^
C:\projects\Arduino\libraries\ESP32DMASPI/ESP32DMASPISlave.h:50:5: sorry, unimplemented: non-trivial designated initializers not supported
C:\projects\Arduino\libraries\ESP32DMASPI/ESP32DMASPISlave.h:53:19: error: 'SPI_DMA_CH_AUTO' was not declared in this scope
     int dma_chan {SPI_DMA_CH_AUTO};  // must be 1, 2 or SPI_DMA_CH_AUTO
                   ^
C:\projects\Arduino\libraries\ESP32DMASPI/ESP32DMASPISlave.h:53:34: error: cannot convert '<brace-enclosed initializer list>' to 'int' in initialization
     int dma_chan {SPI_DMA_CH_AUTO};  // must be 1, 2 or SPI_DMA_CH_AUTO
                                  ^
C:\projects\Arduino\libraries\ESP32DMASPI/ESP32DMASPISlave.h: In member function 'void arduino::esp32::spi::dma::Slave::setDMAChannel(uint8_t)':
C:\projects\Arduino\libraries\ESP32DMASPI/ESP32DMASPISlave.h:188:24: error: 'SPI_DMA_CH_AUTO' was not declared in this scope
             dma_chan = SPI_DMA_CH_AUTO;
                        ^
Using library ESP32DMASPI at version 0.2.0 in folder: C:\projects\Arduino\libraries\ESP32DMASPI 

```

hideakitai, 2022-01-06:
@JBSchueler 
v0.2.0 requires ESP32 version >= 2.0.0 for Arduino IDE, and not supported on PlatformIO now. Added supported version matrix to README

JBSchueler, 2022-01-06:
@hideakitai
Thanks for the information, I was not aware of that.

closed 2022-01-06 by JBSchueler