hideakitai/ESP32DMASPI #16: queue : idle time between 2 transactions

JBSchueler, 2022-01-05:
What is the  minimal idle time between 2 queue items.
In other words, what is the time between an end one transfer and program/start the next one of the queue.

Every 125 microseconds I get a new transfer of 4 bytes at 500kHz transfer rate (not able to change that figure). If I queue for 16 or 32 or 64 transfers, it doesn't matter, I always loses every even transfer transfer. So it looks like it takes more than 62.5 microseconds.
Has someone verifies this?

I only have 62.5 us to end a queue and to start the next one.
Even if I preset multiple queues, e.g. 4, and when one queue is ready I initialize the next one so always at least 3 queues are in line.... It always misses every even transfer.

Has someone similar issues?
How can this be solved (if it is solvable)?

hideakitai, 2022-01-06:
Where do you lose the data? Master or slave?

I don't know how you connect your device(s), but maybe related to this one? 
https://github.com/hideakitai/ESP32DMASPI#restrictions-and-known-issues-for-spi-with-dma-buffer-help-wanted

I don't have enough time to analyze these issues, but it depends on the esp-idf version (the internal process of esp32). Arduino IDE + ESP32 v1.0.6 + ESP32DMASPI v0.1.2 works fine, but Arduino IDE + ESP32 v2.0.0 + ESP32DMASPI v0.12 doesn't work correctly. Because of the timing issues.

Also refer to the documents about timing considerations.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html#transfer-speed-considerations
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_slave.html#speed-and-timing-considerations

JBSchueler, 2022-01-06:
I have a I2S source sending dual 32bit samples at 8kHz.
This results in a SPI clock of 512kHz.

I am trying to capture one channel of the stream via SPI. (The two I2S units are already used).
Current state is that it captures the data (4 bytes) using the queue function.

But somehow not every queue is successful and does not return data.
I only check `available()` for received bytes.
When the available bytes is below a threshold I add new queues to have of the maximum allowed queues.
At a certain moment in time I get a warning all queue space is used and the ESP32 resets.

The only root cause I can think of is queue that received less than 4 bytes, but at least 1, are still in the queue space and not released. This I have to check in the spi dma driver of espressif.

I tried to install Arduino IDE 2.0 + ESP32 v2.0.0 + ESP32DMASPI v0.2.0 but I am not successful in installing board ESP32. It doesn't show up in the board list.

JBSchueler, 2022-01-06:
I got it to work with Arduino IDE 1.8.19 + ESP32 v2.0.0 + ESP32DMASPI v0.2.0
Will check later if I still have the same issues or that those have been solved.

JBSchueler, 2022-01-09:
Issue is not the minimal idle time but incorrect settings.
Has been solved by understanding what `.available()`, `.remained()` and `.size()`stands for.

See #15

closed 2022-01-09 by JBSchueler