hideakitai/ESP32DMASPI #17: esp32dmaspi 2.0.0 won't compile on linux with arduino 1.8.19

emard, 2022-01-05:
Arduino 1.8.19 downloaded and installed,
with manage libraries installed ESP32DMASPI v0.2.0
and loaded example "master_simple"

it compiles with v0.1.2 but with v0.2.0 compile
fails with some error:

In file included from /home/davor/Arduino/esp32btgps/sdcard.cpp:22:0:
/home/davor/Arduino/libraries/ESP32DMASPI/ESP32DMASPIMaster.h:51:5: sorry, unimplemented: non-trivial designated initializers not supported
     };
     ^
/home/davor/Arduino/libraries/ESP32DMASPI/ESP32DMASPIMaster.h:51:5: sorry, unimplemented: non-trivial designated initializers not supported
/home/davor/Arduino/libraries/ESP32DMASPI/ESP32DMASPIMaster.h:55:19: error: 'SPI_DMA_CH_AUTO' was not declared in this scope
     int dma_chan {SPI_DMA_CH_AUTO};  // must be 1, 2 or SPI_DMA_CH_AUTO
                   ^
/home/davor/Arduino/libraries/ESP32DMASPI/ESP32DMASPIMaster.h:55:34: error: cannot convert '<brace-enclosed initializer list>' to 'int' in initialization
     int dma_chan {SPI_DMA_CH_AUTO};  // must be 1, 2 or SPI_DMA_CH_AUTO
                                  ^
/home/davor/Arduino/libraries/ESP32DMASPI/ESP32DMASPIMaster.h: In member function 'void arduino::esp32::spi::dma::Master::setDMAChannel(uint8_t)':
/home/davor/Arduino/libraries/ESP32DMASPI/ESP32DMASPIMaster.h:291:24: error: 'SPI_DMA_CH_AUTO' was not declared in this scope
             dma_chan = SPI_DMA_CH_AUTO;
                        ^
/home/davor/Arduino/libraries/ESP32DMASPI/ESP32DMASPIMaster.h: In member function 'void arduino::esp32::spi::dma::Master::addTransaction(uint8_t, uint8_t, uint8_t, uint32_t, uint16_t, uint64_t, size_t, const uint8_t*, uint8_t*)':
/home/davor/Arduino/libraries/ESP32DMASPI/ESP32DMASPIMaster.h:332:43: error: 'SPI_TRANS_VARIABLE_DUMMY' was not declared in this scope
         transactions.back().base.flags |= SPI_TRANS_VARIABLE_DUMMY;
                                           ^
/home/davor/Arduino/libraries/ESP32DMASPI/ESP32DMASPIMaster.h:344:29: error: '__gnu_cxx::__alloc_traits<std::allocator<spi_transaction_ext_t> >::value_type {aka struct spi_transaction_ext_t}' has no member named 'dummy_bits'
         transactions.back().dummy_bits = dummy_bits;

JBSchueler, 2022-01-05:
There are already tickets open for this.
Please read #13 and #14

hideakitai, 2022-01-06:
@emard 
Hi, v0.2.0 requires ESP32 version >= 2.0.0 for Arduino IDE, and not supported on PlatformIO now. Added supported version matrix to README

closed 2022-01-06 by hideakitai