hideakitai/ESP32DMASPI #18: Reading the second message on the queue

sonalsustainable, 2022-01-09:
I am sending 5 bytes and I receive 2 times 5 bytes inside 2 CS actions (as described better on the attached picture)
this is the characteristic of the IC I am communicating. and I am able to read the first 5 bytes of the response from slave but the second 5 byte array I am not able to read with the queue option.

On this picture the part one is easily readable with the ESP32DMASPI Slave library but the second part I wasn't able to read it.

[image]

sonalsustainable, 2022-01-09:
Here is my code:

BufferSize is 5
void task_process_buffer(void* pvParameters) {

  while (1) {
    ulTaskNotifyTake(pdTRUE, portMAX_DELAY);
    //slave.yield();
    // show received data
    for (size_t i = 0; i < BUFFER_SIZE; ++i)
      printf("%X ", spi_slave_rx_buf[i]);
    printf("\n");

    Serial.print("Available: ");Serial.print(slave.available());
    Serial.print(" Remained: ");Serial.print(slave.remained());
    slave.pop();

    xTaskNotifyGive(task_handle_wait_spi);
  }
}

hideakitai, 2022-01-09:
How about reading without tasks (pollling)? Related to #15 ?

hideakitai, 2022-03-09:
First, please consider using this library with 4 bytes aligned buffer. I am closing this issue because no further information has been provided.

closed 2022-03-09 by hideakitai