hideakitai/ESP32DMASPI #19: ESP32DMASPISlave.h:50:5: sorry, unimplemented: non-trivial designated initializers not supported

sonalsustainable, 2022-01-22:
Hello
Before I was using your code without any problem but with you latest code I am getting this errors. 
Thank you for the help

/Arduino/libraries/ESP32DMASPI-master/ESP32DMASPISlave.h:50:5: sorry, unimplemented: non-trivial designated initializers not supported
     };
     ^
/home/peri/Arduino/libraries/ESP32DMASPI-master/ESP32DMASPISlave.h:50:5: sorry, unimplemented: non-trivial designated initializers not supported
/home/peri/Arduino/libraries/ESP32DMASPI-master/ESP32DMASPISlave.h:53:19: error: 'SPI_DMA_CH_AUTO' was not declared in this scope
     int dma_chan {SPI_DMA_CH_AUTO};  // must be 1, 2 or SPI_DMA_CH_AUTO

JBSchueler, 2022-01-25:
Had the same issue as reported 3 week go.
Look at #13 how to solve this issue

jlarnal, 2022-02-01:
"Designated initializers" are a legacy from C and are deemed invalid in C++11. 
They came back in C++20 under the guise of "aggregate initializers", but the fields appearing between the braces must be in the same order as the one they are declared in.

My two cents : do NOT use these kind of initialization patterns, for portability's sake. 😕