hideakitai/ESP32DMASPI #22: non-trivial-designated-initializers-not-supported

DPWilde, 2022-03-23:
Above error when compiling Master_Slave task example in Arduino IDE 2.0.0-rc5 (also in 1.8.19).
This can be resolved by changing the initialisation of bus_cfg as below in both ESP32DMASPISlave.h and ESP32DMASPIMaster.h :

    spi_bus_config_t bus_cfg {
        .mosi_io_num = 13,        // HSPI
        .miso_io_num = 12,        // HSPI
        .sclk_io_num = 14,        // HSPI
		.quadwp_io_num = -1,	  // not used
		.quadhd_io_num = -1,	  // not used
        .max_transfer_sz = 4092,  // default: 4092 if DMA enabled, SOC_SPI_MAXIMUM_BUFFER_SIZE if DMA disabled
        .flags = SPICOMMON_BUSFLAG_SLAVE,
		.intr_flags = 0
    };

In addition another error is that SPI_DMA_CHA_AUTO is no longer valid in the latest ESP32 core ( I have version 1.06).  I resolved by initialising to DMA channel 1 however it would be better to check if free first.

hideakitai, 2022-03-23:
Please see #13

hideakitai, 2022-03-23:
Your esp board firmware version should be >= 2.0.0. Not IDE itself

hideakitai, 2023-08-01:
I'm closing this issue because of no activity.

closed 2023-08-01 by hideakitai