hideakitai/ESP32DMASPI #23: Master compile error

xygax, 2022-05-05:
In file included from C:\Users\STEVES~1\AppData\Local\Temp\arduino_modified_sketch_782353\master_simple.ino:3:0:
\\192.168.31.24\XygaX-Room-101\Embedded\Arduino\libraries\ESP32DMASPI/ESP32DMASPIMaster.h: In member function 'void arduino::esp32::spi::dma::Master::addTransaction(uint8_t, uint8_t, uint8_t, uint32_t, uint16_t, uint64_t, size_t, const uint8_t*, uint8_t*)':
\\192.168.31.24\XygaX-Room-101\Embedded\Arduino\libraries\ESP32DMASPI/ESP32DMASPIMaster.h:350:43: error: 'SPI_TRANS_VARIABLE_DUMMY' was not declared in this scope
         transactions.back().base.flags |= SPI_TRANS_VARIABLE_DUMMY;
                                           ^
\\192.168.31.24\XygaX-Room-101\Embedded\Arduino\libraries\ESP32DMASPI/ESP32DMASPIMaster.h:362:29: error: '__gnu_cxx::__alloc_traits<std::allocator<spi_transaction_ext_t> >::value_type {aka struct spi_transaction_ext_t}' has no member named 'dummy_bits'
         transactions.back().dummy_bits = dummy_bits;
                             ^
exit status 1
Error compiling for board ESP32 Dev Module.

I have tried to add definition for dummy_bits  prior to Begin() however still gives this error

hideakitai, 2023-08-01:
This library is not optimized for current latest ESP32 core. Please let me know:

- IDE (Arduino IDE, PlatformIO, etc.) : 
- Board Firmware Version (see board manager): 
- Library Version of ESP32DMASPI:

hideakitai, 2023-08-14:
The latest arduino-esp32 (v2.0.11) was supported by https://github.com/hideakitai/ESP32DMASPI/pull/31. Please feel free to reopen this issue if you still need help.

closed 2023-08-14 by hideakitai