hideakitai/ESP32DMASPI #34: received buffer on ESP32 (slave) is repeating the first 256 bytes

hasanayan, 2023-11-01:
First, thanks for your work!

I'm using Teensy4.1 in master mode and Nodemcu-32s (ESP-WROOM-32)  in slave mode. I am trying to send 8292 bytes from teensy and reading that from esp32. ESP32's receive buffer is repeating the first 256 bytes received. Could you please show me what I am doing wrong?

Code on Teensy4.1
`
#include "Arduino.h"
#include <TsyDMASPI.h>

#define CS_PIN 10
#define DMASIZE 8192
uint8_t src[DMASIZE];
volatile uint8_t dest[DMASIZE];
volatile uint8_t dest1[DMASIZE];

void setup()
{
  Serial.begin(115200);

  TsyDMASPI0.begin(CS_PIN, SPISettings(2000000, MSBFIRST, SPI_MODE1));
}

void loop()
{
  for (int i = 0; i < DMASIZE; i++)
  {
    src[i] = i;
  }

  TsyDMASPI0.transfer(src, dest, DMASIZE);

  delay(4000);
}
`

Code running on nodemcu-32s:

`
#include <Arduino.h>
#include <ESP32DMASPISlave.h>

#define CS_PIN 5
#define MISO 19
#define MOSI 23
#define SCK 18

ESP32DMASPI::Slave slave;

static const uint32_t BUFFER_SIZE = 8192;
uint8_t *spi_slave_tx_buf;
uint8_t *spi_slave_rx_buf;

void setup()
{
    Serial.begin(115200);

    // to use DMA buffer, use these methods to allocate buffer
    spi_slave_tx_buf = slave.allocDMABuffer(BUFFER_SIZE);
    spi_slave_rx_buf = slave.allocDMABuffer(BUFFER_SIZE);

    slave.setDataMode(SPI_MODE1);
    slave.setMaxTransferSize(BUFFER_SIZE);

    slave.begin(HSPI, SCK, MISO, MOSI, SS);
}

void loop()
{

    slave.wait(spi_slave_rx_buf, spi_slave_tx_buf, BUFFER_SIZE);

    while (slave.available())
    {
        Serial.println("data received");

        for (int i = 0; i < BUFFER_SIZE; i++)
        {
            Serial.printf("%i: %i\n", i, spi_slave_rx_buf[i]);
        }

        slave.pop();
    }
}

`

The output I get on ESP32 is

data received
0: 0
1: 1
2: 2
3: 3
4: 4
5: 5
6: 6
7: 7
8: 8
9: 9
10: 10
11: 11
12: 12
13: 13
...
...
...
7162: 250
7163: 251
7164: 252
7165: 253
7166: 254
7167: 255
7168: 0
7169: 1
7170: 2
7171: 3
7172: 4
7173: 5
7174: 6
...
...
...
8178: 242
8179: 243
8180: 244
8181: 245
8182: 246
8183: 247
8184: 248
8185: 249
8186: 250
8187: 251
8188: 252
8189: 253
8190: 254
8191: 255

hasanayan, 2023-11-01:
oops this is embarassing, I understood my mistake. uint8_t can encode up to 255 :D

hideakitai, 2023-11-01:
If you have solved the problem, please close this issue :)

closed 2023-11-03 by hideakitai