hideakitai/ESP32DMASPI #37: Trying to read device ID of ADS1299 but getting 0

MonisRaza123, 2024-01-03:
I have designed a board around ADS1299. I know the board works fine as I have implemented normal SPI (without DMA) which is working as expected. When using this library to implement DMA based SPI and trying to read device ID I'm getting zero as output which shouldn't be the case. 

I am using 0.1.2 version of this library with ARDUINO 2.0.3 as the IDE 

Here is the code that I've written-
`#include <ESP32DMASPIMaster.h>
#define ID 0x00
#define CONFIG1 0x01
#define CONFIG2 0x02
#define CONFIG3 0x03
#define LOFF 0x04
#define CH1SET 0x05
#define CH2SET 0x06
#define CH3SET 0x07
#define CH4SET 0x08
#define CH5SET 0x09
#define CH6SET 0x0A
#define CH7SET 0x0B
#define CH8SET 0x0C
#define BIAS_SENSP 0x0D
#define BIAS_SENSN 0x0E
#define LOFF_SENSP 0x0F
#define LOFF_SENSN 0x10
#define LOFF_FLIP 0x11
#define LOFF_STATP 0x12
#define LOFF_STATN 0x13
#define GPIO 0x14
#define MISC1 0x15
#define MISC2 0x16
#define CONFIG4 0x17

#define _WAKEUP 0x02 
#define _STANDBY 0x04 
#define _RESET 0x06
#define _START 0x08 
#define _STOP 0x0A
#define _RDATAC 0x10 
#define _SDATAC 0x11 
#define _RDATA 0x12 

#define _RREG 0x20 
#define _WREG 0x40 

#define HSPI_MISO   12
#define HSPI_MOSI   13
#define HSPI_SCLK   14
#define HSPI_SS     15

#define CLKSEL 32
#define DRDY 34
#define DAISY_IN 33
#define START 25
#define RESET 26
#define PWDN 27

float tCLK=4.8828125e-7;
static const int spiClk = 2000000; // 1 MHz

static const int BUFFER_SIZE = 32;
uint8_t* spiTxBuf;
uint8_t* spiRxBuf;

ESP32DMASPI::Master master;

void setup() {
  Serial.begin(115200);
  Serial.flush();
  delay(1000);
  while(!Serial);
  pinMode(HSPI_SCLK, OUTPUT);
  pinMode(HSPI_MOSI, OUTPUT);
  pinMode(HSPI_MISO, INPUT);
  pinMode(HSPI_SS, OUTPUT);

  digitalWrite(HSPI_SCLK, LOW);
  digitalWrite(HSPI_MOSI, LOW);
  digitalWrite(HSPI_SS, HIGH);

  spiTxBuf = master.allocDMABuffer(BUFFER_SIZE); 
  spiRxBuf = master.allocDMABuffer(BUFFER_SIZE);

  
  master.setDataMode(SPI_MODE1);
  master.setFrequency(2000000); //2 MHz
  master.setMaxTransferSize(BUFFER_SIZE);
  master.setDMAChannel(1);
  master.setQueueSize(1); 
  master.begin(HSPI,14,12,13,15); 
  

  
  pinMode(RESET, OUTPUT);
  digitalWrite(RESET, HIGH);
  pinMode(PWDN, OUTPUT);
  digitalWrite(PWDN, HIGH);
  pinMode(START, OUTPUT);
  pinMode(DRDY, INPUT);
  pinMode(CLKSEL, OUTPUT);
  pinMode(DAISY_IN, OUTPUT);

  delay(128);
  
  digitalWrite(CLKSEL, HIGH);
  delayMicroseconds(20);
  digitalWrite(RESET,LOW);
  delayMicroseconds(1);
  digitalWrite(RESET, HIGH);
  delayMicroseconds(9);

  SDATAC();
  getDeviceID();
}

void loop() {

}

void getDeviceID() {

   //Construct SPI message
  spiTxBuf[0] = _RREG; 
  spiTxBuf[1] = 0x00;
  spiTxBuf[2] = 0x00; 

  digitalWrite(HSPI_SS, LOW); 
  delayMicroseconds(20);

  master.transfer(spiTxBuf, spiRxBuf, 3); 

  delayMicroseconds(20);
  digitalWrite(HSPI_SS, HIGH);

  Serial.print(spiRxBuf[2], BIN);
  

}

void SDATAC(){
  
  spiTxBuf[0] = _SDATAC;
  digitalWrite(HSPI_SS, LOW); 
  delayMicroseconds(20);
  master.transfer(spiTxBuf, 1); 
  delayMicroseconds(20);
  digitalWrite(HSPI_SS, HIGH); 
}`

Here is the output:
[image]

hideakitai, 2024-01-04:
DMA buffer size on `transfer()` should be multiples of 4 bytes. And I wonder if ADS1299 allows such a transaction. Please try again with the latest library version. Also, I think there's no need to use DMA in such a small transaction (a standard SPI transaction is enough for such usage)

MonisRaza123, 2024-01-06:
Thank you for your response, it helped us resolve the issue. I know that normal SPI is more than capable of conducting this transfer, was testing out DMA for implementation of ping pong buffer with a much bigger chunk of data that has to be sent to a remote database in close to real time.

closed 2024-01-06 by MonisRaza123