hideakitai/ESP32DMASPI #38: SPI Ping-Pong DMA

MonisRaza123, 2024-01-07:
I am trying to conduct SPI-based communication with ADS1299 using DMA and implement PING PONG buffer to send data continuously to a remote database. 
This is what I'm trying to implement:
[image]
[image]

I have made two receive buffers of 2700 bytes each along with a single transmit buffer. The data from ADC is sampled at 16kSPS, and the size of the buffer as stated has been chosen keeping in consideration the time taken to process a single request by the database.

I can store the data alternatively between the two buffers. I am doing so by simply checking the index of the buffer that is currently being written to serve as the point of action to change the receiving buffer.

I want to know whether in your library any interrupt is raised once the receiving buffer gets full.

hideakitai, 2024-01-09:
When the transaction from master to slave has finished, this function is called. But it is not exposed to the user. 

https://github.com/hideakitai/ESP32DMASPI/blob/c71071b0a5bc318a035161bb47d7a6a732c47410/ESP32DMASPISlave.h#L251-L255

MonisRaza123, 2024-01-10:
I know this is beyond the scope of the support for this library but could you help me implement the above logic as specified in my comment? I'm not expecting you to write the entire code for me just guide me through the flow of the program (pseudocode) to achieve my objective. That would be of great help!

hideakitai, 2024-01-11:
Does that mean you would like to commission us as a job? If so, of course it is possible.

MonisRaza123, 2024-01-13:
Haha I wish I could but I'm just an undergrad student and cannot afford to do so. Regardless can you tell me why my core gets dumped when trying to call master.transfer() inside an ISR? According to my undrestanding DMA based transaction shouldn't be costing CPU cycles then why is the core getting dumped?

hideakitai, 2024-01-14:
transfer() blocks task. ISR does not allow such a blocking function. If you want to transfer soon after ISR is called, 

- create a new task that calls transfer() if triggered
- trigger transfer() from ISR (for example xTaskNotifyFromISR())
- task wait for the next trigger

manyabansal, 2024-01-21:
Hi! I'm trying to perform the same project that @MonisRaza123 mentioned. 
This time, instead of using transfer() function, we tried to use the queue() function to perform the same task. 
I'm queueing the task inside the ISR, and setting a flag. And, in the main loop, under the flag condition, I'm trying to get the result using master.yield() .
However, I keep getting the following error:

**abort() was called at PC 0x40084fd3 on core 1
Backtrace: 0x400838d1:0x3ffbf34c |<-CORRUPTED**

I want to know what is causing this error, and how can it be solved?

hideakitai, 2024-01-30:
Could you try v0.4.0 release? Also notice [Notes for Communication Errors](https://github.com/hideakitai/ESP32DMASPI?tab=readme-ov-file#notes-for-communication-errors)

hideakitai, 2024-02-12:
If you still need help, please feel free to reopen this issue.

closed 2024-02-12 by hideakitai