hideakitai/ESP32DMASPI #40: Lost byte's

JoeReturn, 2024-01-21:
Dear Hideaki Tai,

I have been trying to use the ESP32 library you created for some time, but it doesn't seem to work properly. I am attaching a source code in which there is both TCP and SPI slave communication.

The issue is that, in the master SPI device, the first 168 bytes (from 4108) are consistently missing in a loop, and I don't really understand what kind of error could cause this.

Thank you for your response.

hideakitai, 2024-01-31:
Sorry, could you solve this problem? Please try v0.4.0 release. Also notice [Notes for Communication Errors](https://github.com/hideakitai/ESP32DMASPI?tab=readme-ov-file#notes-for-communication-errors)

closed 2024-01-31 by hideakitai

JoeReturn, 2024-01-31:
Dear Friend,

When i memcpy()  made change the spi _slave_tx_buff  and read only
SlaveRemained... this made qeuer... only

Now this good
Thanks
Have A Nice Day :)

Best Regards
Joe

[image: image.png]

[image: image.png]

Hideaki Tai ***@***.***> ezt írta (időpont: 2024. jan. 31.,
Sze, 2:15):