hideakitai/ESP32DMASPI #53: 1ms minimum packet time.

Artemkamax, 2024-03-27:
Hi! I trying to use example "transfer_big_data_in_the_background"
sending 256bytes at 20Mhz speed from one esp32 to another as fast as i can, but as I see master send packet, and wait somesthing. now packet transfering time is 1ms. If I send 1200bytes, transfering time also 1ms. How I can change it?
I can transfer 16000 bytes in one packet, but slave cant recieve this packet, some bytes are missed. 
I need to reach at least 2 160 000Bytes/s
teoretical at 20Mhz I can 2.5Mbytes/s and 5Mbytes/s at 40Mhz, but have problems

hideakitai, 2024-03-27:
It's probably FreeRTOS context switch. For more speed, we recommend writing your own code without this library.

hideakitai, 2024-03-28:
> 1200bytes in a packet at 20mhz is maximum?

No. It may be a problem of hardware (noise, etc.)

hideakitai, 2024-04-09:
If you still need help, please reopen this issue.

closed 2024-04-09 by hideakitai