hideakitai/ESP32DMASPI #56: Cannot Receive Transaction over 32768 bytes

grahamgrieb, 2024-05-31:
If my SPI transfer is over 32768 bytes, it seems like the receive size I get back will always be the size of the transfer minus 32,768 (if I do a 50000 byte transfer, my received_bytes will be 17232). My buffer size is well over 32,768 so I don't think that is the issue. I feel like it might be an overflow issue somewhere since 32,768 is the max value of a signed 16-bit integer. My SPI device i is sending data at 2 MHz. I've attached my code below. I am running on an ESP32-S3 on PlatformIO. 

```#include <ESP32DMASPISlave.h>

ESP32DMASPI::Slave slave;
static constexpr size_t BUFFER_SIZE = 90000;
static constexpr size_t QUEUE_SIZE = 1;
uint8_t *dma_rx_buf;

void loop()
{
  Serial.println("Waiting for data");
  //slave.queue(NULL, dma_rx_buf, BUFFER_SIZE);
  //const std::vector<size_t> received_bytes = slave.wait();
  size_t received_bytes = slave.transfer(NULL, dma_rx_buf, BUFFER_SIZE);
  Serial.println("Data received");
  Serial.println(received_bytes);
  //print error
  
  Serial.println(slave.errors()[0]);

}

void setup () {  
  Serial.begin(9600);
  
  //dma_tx_buf = slave.allocDMABuffer(BUFFER_SIZE);
  dma_rx_buf = slave.allocDMABuffer(BUFFER_SIZE);

  slave.setDataMode(SPI_MODE3);           // default: SPI_MODE0
  slave.setMaxTransferSize(BUFFER_SIZE);  // default: 4092 bytes
  slave.setQueueSize(QUEUE_SIZE);         // default: 1
  
  
  // begin() after setting
  slave.begin();  // default: HSPI (please refer README for pin assignments)

}

hideakitai, 2024-06-01:
Could you try again with `-DCORE_DEBUG_LEVEL=4` to show the debug log? After that, please send me the whole log output.

https://community.platformio.org/t/how-to-set-up-log-level-to-be-able-to-debug-the-esp32/8278

grahamgrieb, 2024-06-01:
This is what it looked like for a 50000 byte transfer. It says the receive size is 17232 (50000 minus 32768).

[D][ESP32DMASPISlave.h:112] spi_slave_task(): [ESP32DMASPISlave] spi_slavve_task start
Waiting for data
[   324][D][ESP32DMASPISlave.h:140] spi_slave_task(): [ESP32DMASPISlave] new transaction request received (size = 1)
[  3073][D][ESP32DMASPISlave.h:166] spi_slave_task(): [ESP32DMASPISlave] transaction complete: 137856 bits (17232 bytes) received
[  3075][D][ESP32DMASPISlave.h:189] spi_slave_task(): [ESP32DMASPISlave] all requested transactions completed
Data received
17232

hideakitai, 2024-06-01:
Hmm, then ESP32's API does not receive more than 32768 bytes. The API does not limit the transfer size to it (and in my library). I'd like to confirm if the master really sends more than 32768 bytes. It is wrapped by signed 16 bytes (int16_t). Does the master can send bytes more than the max of int16_t?

grahamgrieb, 2024-06-01:
I am using a logic analyzer to view the SPI messages so I can confirm it is actually 50000 bytes. It might just be an ESP32 API issue then.

hideakitai, 2024-06-01:
Yes, maybe limited by esp-idf for ESP32-S3 https://github.com/esp-rs/esp-idf-hal/issues/377

hideakitai, 2024-06-01:
Closing this issue because this library does not cause it

closed 2024-06-01 by hideakitai