hideakitai/ESP32DMASPI #62: Is there support for ESP8266/RTL8720/raspberry/orangepi ?

PospelovDaniil, 2024-06-26:
Is there support for ESP8266/RTL8720/raspberry/orangepi ?

Question, is it possible to **trivially** connect(control)
ESP32 using SPI and this library with other microcontroller models or single board computers ?

When I said control, I meant:

0) The controlling microcontroller sent a "command" to the subordinate one
1) The slave received the command, started executing it
2) It took some time to execute the command
3) The controlling microcontroller determined somehow (how, by the way?),
that the subordinate has executed the command and is ready to send the result of the command to the master.

Simple example:
`enum Commands { GetSum, GetMultiple /*etc....*/ }; // contained in both the master and slave microcontrollers`

0) Master sent bytes: {Commands::GetSum,2,2}
1) The slave received the command, realized that it was necessary to add the sum, 2 and 2
2) The subordinate added 2 and 2 for some time (even though in this example it was very fast, but still some time).
3) The master controller somehow realized that the result was ready and got it - 4 (while the calculation was going on, the microcontroller was waiting, synchronously)

My project idea:
A simple and fast 2.4ghz airwaves congestion analyzer.
Connect 3 ESP32 via SPI to start with [listening to channels 1,6,11](https://www.metageek.com/training/resources/why-channels-1-6-11/).
with the 4th ESP32 and analyze the load of WIFI air to visualize it on the screen/web interface.
I think with this library I can do it.

Very simplistically the algorithm will be like this:
The main ESP32 polls synchronously (for ease of understanding the logic and code)
each of the microcontrollers, what is the current load on the air and then outputs this data somewhere.

And if I replace 3 esp32 with 3 esp8266 to make the project cheaper, I can do it with this library ?
And if I want to add a 5Ghz RTL8720 module, how do I communicate with it ?
And if I want to replace the control esp32 with my orangepi5 ?

I am not a professional developer for microcontrollers and my question might have been stupid, sorry about that.
If you have simple projects with similar data transfer over SPI you can share them.

hideakitai, 2024-06-26:
This library is available only for the ESP32 series. I don't know about RTL8720, but you can communicate with any other MCU/devices with SPI.

Please see this example for the polling scenario

https://github.com/hideakitai/ESP32DMASPI/tree/main/examples/transfer_big_data_in_the_background

PospelovDaniil, 2024-06-26:
All right, thanks for the answer.

closed 2024-06-26 by PospelovDaniil