hideakitai/ESP32DMASPI #64: My program is executing "master.begin();" crashed

piul200, 2024-08-15:
[image]

ELF file SHA256: 8e95fda5b922372e

Re-entered core dump! Exception happened during core dump!
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4202231e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (spi_master_task)
Core  1 register dump:
PC      : 0x4037ce53  PS      : 0x00060936  A0      : 0x8037ad0b  A1      : 0x3fcec240
A2      : 0x3fcf0810  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060923
A6      : 0x00060920  A7      : 0x0000abab  A8      : 0xb33fffff  A9      : 0xffffffff
A10     : 0x00060923  A11     : 0x00000000  A12     : 0x00060920  A13     : 0x0000000c
A14     : 0x02cf0810  A15     : 0x00ffffff  SAR     : 0x0000001f  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000

Backtrace: 0x4037ce50:0x3fcec240 0x4037ad08:0x3fcec280 0x4037adc0:0x3fcec2c0 0x4037867d:0x3fcec2e0 0x40378755:0x3fcec310 0x40378824:0x3fcec330 0x4201887a:0x3fcec360 0x4201b8a1:0x3fcec670 0x420235f5:0x3fcec6a0 0x4200dfbe:0x3fcec6d0 0x4200e053:0x3fcec700 0x4200352e:0x3fcec780 0x4200418d:0x3fcec830 0x42001d1b:0x3fcec880

hideakitai, 2024-08-15:
What is the version of Arduino-esp32?

Can you run the example as is without crashing? For example https://github.com/hideakitai/ESP32DMASPI/blob/main/examples/transfer_big_data_one_by_one/transfer_big_data_one_by_one_master/transfer_big_data_one_by_one_master.ino

piul200, 2024-08-15:
My development environment is vscode+platformio (Core 6.1.15 Home 3.4.4).
I try to run this example, but it still crashes. I removed "master.begin();" and the program didn't crash, but doing so would cause DMASPI to not work

hideakitai, 2024-08-15:
What is the version of esp32 of platformio?

https://docs.platformio.org/en/latest/platforms/espressif32.html

Please try to add `monitor_filters = esp32_exception_decoder` to your `platformio.ini` to decode the backtrace. After running and crashing again, please give me your log again.

piul200, 2024-08-16:
Thanks for your reply, here is my `platformio.ini` and the log of the program crashing again
```
[env:4d_systems_esp32s3_gen4_r8n16]
platform = espressif32
board = 4d_systems_esp32s3_gen4_r8n16
framework = arduino
lib_deps = hideakitai/ESP32DMASPI@^0.6.3
monitor_filters = esp32_exception_decoder
```
[log.txt](https://github.com/user-attachments/files/16633251/log.txt)

hideakitai, 2024-08-16:
Thank you! Could you try again with `build_flags = -DCORE_DEBUG_LEVEL=4` to your `platformio.ini`? I need this flag and debug log to determine the cause of the panic.

piul200, 2024-08-16:
Okay, here's the new log
```
[env:4d_systems_esp32s3_gen4_r8n16]
platform = espressif32
board = 4d_systems_esp32s3_gen4_r8n16
framework = arduino
lib_deps = hideakitai/ESP32DMASPI@^0.6.3
monitor_filters = esp32_exception_decoder
build_flags = -DCORE_DEBUG_LEVEL=4
```
[log_1.txt](https://github.com/user-attachments/files/16633690/log_1.txt)

hideakitai, 2024-08-16:
Can you try the following `begin()` (with arguments) method instead of `begin()` (empty argment)?

```c++
  master.begin(HSPI, SCK, MISO, MOSI, SS);
```

hideakitai, 2024-08-16:
OK, thank you for testing.

Yes, because you are sending the data as fast as possible in the loop() (no delay())

closed 2024-08-16 by piul200