hideakitai/ESP32DMASPI #66: Quad SPI mode

ivantaka, 2025-01-23:
Hi, I'm trying to configure Quad SPI mode using master examples. Starting from queue and wait master example, I only changed the begin() method with the 6 pins for QSPI; and the queue size to 1. 
`master.begin(VSPI, SCK, SS, MOSI, MISO, 22, 21);`

It seems to be sending data only to MOSI pin (data0). MISO pin (data1) is constant low and pins WD (data2) and HD (data3) are constant high. 

Did I miss some configuration or method to transfer data in QSPI?

Thanks in advance.

hideakitai, 2025-02-02:
@ivantaka [v0.6.5](https://github.com/hideakitai/ESP32DMASPI/releases/tag/v0.6.5) fixes this issue. Could you confirm if the problem has fixed?

ivantaka, 2025-02-04:
Hi, sorry for the delay and thanks for the reply. 

v0.6.5 definitely helped, I just had to also set Device Flags with half-duplex mode. But it seems to be only outputting data to io0 and io1; io2 and io3 are always high. I tried other pins with no success.

This is the example code I'm using, I'm just sending data and checking if I get any changes on the pins:

```
#include "ESP32DMASPIMaster.h"
#include "helper.h"

ESP32DMASPI::Master master;

static constexpr size_t BUFFER_SIZE = 256; // should be multiple of 4
static constexpr size_t QUEUE_SIZE = 1;
uint8_t *dma_tx_buf;

void setup() {
  Serial.begin(115200);
  
  delay(2000);

  /******  MASTER SIDE *******/
  // to use DMA buffer, use these methods to allocate buffer
  dma_tx_buf = master.allocDMABuffer(BUFFER_SIZE);
  
  master.setDataMode(SPI_MODE0);           // default: SPI_MODE0
  master.setFrequency(1000000);            // default: 8MHz (too fast for bread board...)
  master.setMaxTransferSize(BUFFER_SIZE);  // default: 4092 bytes
  master.setQueueSize(QUEUE_SIZE);         // default: 1
  master.setDeviceFlags(SPI_DEVICE_HALFDUPLEX);
  
  // begin() after setting
//  if(!master.begin(VSPI)) Serial.println("Failed master initialization.");
//  master.begin(VSPI, SCK, SS, MOSI, MISO);  // default: HSPI (please refer README for pin assignments)
  master.begin(VSPI, SCK, SS, MOSI, MISO, 17, 16);  // default: HSPI (please refer README for pin assignments)

  delay(2000);
}

void loop() {
  initializeBuffers(dma_tx_buf, NULL, BUFFER_SIZE);
  
//  master.queue(0, 0, 0, 0, 0, 0, dma_tx_buf, NULL, BUFFER_SIZE);
  master.queue(dma_tx_buf, NULL, BUFFER_SIZE);

  Serial.println("wait for the completion of the queued transactions...");
  master.wait();

  // process received data from slave
  Serial.println("all queued transactions completed. start verifying received data from master");

  // verify and dump difference with received data
  // NOTE: we need only 1st results (received_bytes[0])
//  if (verifyAndDumpDifference("slave", dma_tx_buf, BUFFER_SIZE, "master", dma_rx_buf, received_bytes[0])) {
//      Serial.println("successfully received expected data from master");
//  } else {
//      Serial.println("Unexpected difference found between master/slave data");
//  }
}
```

I'm using a low quality hand-held oscilloscope so I can't really trigger well to read what the actual data is. I will reply if I find something.

ivantaka, 2025-02-04:
Hi, 

I've managed to analyze the data by sending only 4 bytes. It seems to be sending the data using only data_io0 and data_io1, because it's sending on 16 sck cycles.

ivantaka, 2025-02-04:
Hi again,

Finally I managed to make it work here. It seems that in the generateTransaction() function, the code was skipping the if condition for QIO, setting the flag manually seems to work. I might have configured or initialized something wrong or maybe a syntax problem on my side. 

Anyways, thanks for the attention.

closed 2025-02-04 by ivantaka