hideakitai/ESP32DMASPI #71: reset when using you sample code in ESP32C3

git-78, 2025-02-20:
Thank you for your library. It looks very interesting but, pitifully, even when reusing your sample code on my ESP32C3 (Seeed Studion Xiao), the device keeps resetting itself in the setup method. In the latter, I've tried various calls to `begin` method (all 3 should be actually the same) and all are crashing at this step. Does anyone has an idea on this? Thank you!

My code:
```
#define PIN_MOSI 10 // by default
#define PIN_MISO 9 // by default
#define PIN_CLK 8 // by default
#define PIN_CS 20 // by default

#include <ESP32DMASPISlave.h>
#include "helper.h"

ESP32DMASPI::Slave slave;

static constexpr size_t BUFFER_SIZE = 256; // should be multiple of 4
static constexpr size_t QUEUE_SIZE = 1;
uint8_t *dma_tx_buf;
uint8_t *dma_rx_buf;

void print(const char* name, int pin) {
  Serial.print(name);
  Serial.println(pin);
}

void setup()
{
    Serial.begin(115200);

    delay(2000);
    
    print("SS:   ", SS);
    print("MOSI: ", MOSI);
    print("MISO: ", MISO);
    print("SCK:  ", SCK);

    // to use DMA buffer, use these methods to allocate buffer
    dma_tx_buf = slave.allocDMABuffer(BUFFER_SIZE);
    dma_rx_buf = slave.allocDMABuffer(BUFFER_SIZE);

    slave.setDataMode(SPI_MODE0);           // default: SPI_MODE0
    slave.setMaxTransferSize(BUFFER_SIZE);  // default: 4092 bytes
    slave.setQueueSize(QUEUE_SIZE);         // default: 1

    // begin() after setting
    //slave.begin();  // default: HSPI (please refer README for pin assignments)
    //slave.begin(HSPI, SCK, MISO, MOSI, SS);
    slave.begin(HSPI, PIN_MOSI, PIN_MISO, PIN_CLK, PIN_CS);

    Serial.println("start spi slave");
}

void loop()
{
}

```

Console output:
```
SS:   20
MOSI: 10
MISO: 9
SCK:  8

assert failed: xTaskCreatePinnedToCore freertos_tasks_c_additions.h:162 (( ( ( ( BaseType_t ) xCoreID ) >= 0 && ( ( BaseType_t ) xCoreID ) < 1 ) ? ( ( BaseType_t ) 1 ) : ( ( BaseType_t ) 0 ) ) == ( (
Core  0 register dump:
MEPC    : 0x40381370  RA      : 0x40385094  SP      : 0x3fc986e0  GP      : 0x3fc8d600  
TP      : 0x3fc989d0  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x0000007e  S1      : 0x3fc988d7  A0      : 0x3fc98744  A1      : 0x3c036b32  
A2      : 0x0000007e  A3      : 0x00000028  A4      : 0x00000001  A5      : 0x3fc90000  
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x00000000  S3      : 0x3fc98730  
S4      : 0x3fc98738  S5      : 0x3c036b98  S6      : 0x00000005  S7      : 0x3fc8e46c  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
3fc986e0: 0x3fc989d0 0x00000000 0x3c03c478 0x4038b730 0x4038613c 0x403848b4 0x3fc98770 0x3c036ab4
3fc98700: 0x3fc989d0 0x00000000 0x4200137e 0x00323631 0x00000001 0x3fc8e048 0x3c03c478 0x3fc8e058
3fc98720: 0x3c036b98 0x3fc8e05c 0x3fc9870c 0x3fc8e060 0x3c036ab4 0x3fc8dce0 0x00000000 0x00000000
3fc98740: 0x3f000000 0x65737361 0x66207472 0x656c6961 0x78203a64 0x6b736154 0x61657243 0x69506574
3fc98760: 0x64656e6e 0x6f436f54 0x66206572 0x72656572 0x5f736f74 0x6b736174 0x5f635f73 0x69646461
3fc98780: 0x6e6f6974 0x3a682e73 0x20323631 0x28202828 0x28202820 0x73614220 0x70795465 0x20745f65
3fc987a0: 0x43782029 0x4965726f 0x20292044 0x30203d3e 0x20262620 0x20282028 0x65736142 0x65707954
3fc987c0: 0x2920745f 0x6f437820 0x44496572 0x3c202920 0x29203120 0x28203f20 0x42202820 0x54657361
3fc987e0: 0x5f657079 0x20292074 0x20292031 0x2028203a 0x61422028 0x79546573 0x745f6570 0x30202920
3fc98800: 0x29202920 0x203d3d20 0x00282028 0x4a5ebfa7 0x00000002 0x3fc8e3e0 0x3fc938b4 0x42000472
3fc98820: 0x00000800 0x3fc8e3e0 0x00000011 0x40387f34 0x0000000f 0x00000011 0x3fc98878 0x00000000
3fc98840: 0x00000000 0x00000000 0x3fc98868 0x3fc90000 0x3fc98868 0x3fc8e3e0 0x00000011 0x4200029e
3fc98860: 0x3fc98868 0x00000002 0x3f003032 0x3fc8d600 0x3fc989d0 0x00000000 0x3fc98880 0x00000000
3fc98880: 0x00000000 0x76616c73 0x61745f65 0x005f6b73 0x3fc938b4 0x00000011 0x0000001e 0x37303630
3fc988a0: 0x39303830 0x31313031 0x33313231 0x35313431 0x37313631 0x39313831 0x31323032 0x33323232
3fc988c0: 0x35323432 0x37323632 0x39323832 0x31333033 0x33333233 0x35333433 0x37333633 0x39333833
3fc988e0: 0x31343034 0x33343234 0x35343434 0x37343634 0x39343834 0x31353035 0x33353235 0x35353435
3fc98900: 0x37353635 0x39353835 0x31363036 0x33363236 0x35363436 0x37363636 0x39363836 0x31373037
3fc98920: 0x33373237 0x35373437 0x37373637 0x39373837 0x31383038 0x33383238 0x35383438 0x37383638
3fc98940: 0x39383838 0x31393039 0x33393239 0x35393439 0x37393639 0x39393839 0x3fc8e000 0x4a5ebfa7
3fc98960: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc90000 0x3fc8e000 0x4200042c
3fc98980: 0x00000000 0x00000000 0x42004000 0x42003caa 0x00000000 0x00000000 0x00000000 0x40386004
3fc989a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc989c0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc989e0: 0xbaad5678 0x00000168 0xabba1234 0x0000015c 0x3fc987e0 0x00000c0f 0x3fc8ea08 0x3fc8ea08
3fc98a00: 0x3fc989f0 0x3fc8ea00 0x00000018 0x00000000 0x00000000 0x3fc989f0 0x00000000 0x00000001
3fc98a20: 0x3fc969e0 0x706f6f6c 0x6b736154 0x00000000 0x00000000 0x3fc989d0 0x00000005 0x00000000
3fc98a40: 0x00000001 0x00000000 0x00000000 0x00000000 0x0000336d 0x00000000 0x3fc905a4 0x3fc9060c
3fc98a60: 0x3fc90674 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x3fc9350c 0x4200c358
3fc98a80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc98aa0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc98ac0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
```

git-78, 2025-02-20:
Found error: in ESP32DMASPIMaster.h (and Slave), xTaskCreatePinnedToCore 's last argument to specify core was 1, but ESP32C3 has only one core. Changing it to tskNO_AFFINITY solve runtime crash.

Yes, will ESP32DMASPI work correctly on one single core, and with current code?

hideakitai, 2025-05-02:
Thank you, fixed in v0.6.8 https://github.com/hideakitai/ESP32DMASPI/pull/78/files

closed 2025-05-02 by hideakitai