hideakitai/ESP32DMASPI #80: Error on example code "spi_master: check_trans_valid: Incompatible when setting to both multi-line mode and half duplex mode"

git-78, 2025-05-22:
Hi,
On my EPS32S3, the non-DMA examples from the ESP32SPISlave lib run fine, but all examples from ESP32DMASPI fail for the master with the error:
E (4011) spi_master: check_trans_valid(1063): Incompatible when setting to both multi-line mode and half duplex mode

I have searched and read a lot , but couldn't find any solution to it. Can anyone help?
Thank you in advance!

hideakitai, 2025-06-01:
Are you using the code in the examples “as is”? None of the codes in the examples directory sets `SPI_DEVICE_HALFDUPLEX`. Please refer to this Espressif documentation for individual settings and set them yourself.

https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/spi_master.html

hideakitai, 2025-06-15:
If you still need help, please reopen this issue

closed 2025-06-15 by hideakitai